hybridgroup/node-bebop #1: dgram 404

braden-smith, 2015-05-07:
Hello,
trying to create sample project in a browser for now.

Your JS has 3 dependencies: net/util/dgram.

"nodejs install dgram" results in a 404 however. Could you clarify where the dgram you used came from?

Thanks

zankich, 2015-05-08:
@braden-smith `dgram` is part of the node.js std lib https://nodejs.org/api/dgram.html.

For use in a browser  there needs to be a `dgram` compatible wrapper library around  something like chrome's udp library https://developer.chrome.com/apps/sockets_udp.

So as it stands right now, there is some extra work which needs to be done to run this in a browser....

deadprogram, 2016-04-28:
Hi @braden-smith did you ever look at https://github.com/feross/chrome-dgram or https://github.com/DecipherCode/firefox-dgram to solve this?