hybridgroup/node-bebop #5: getMjpegStream is undefined

timotius02, 2015-06-21:
I was doing some testing with node-bebop and opencv when I stumbled upon this. I am currently using v0.2.0.
```bh
/Users/timotius02/code/drone/app.js:12
    var mjpg = drone.getMjpegStream();
                     ^
TypeError: undefined is not a function
    at /Users/timotius02/code/drone/app.js:12:22
    at Socket.<anonymous> (/Users/timotius02/code/drone/node_modules/node-bebop/lib/bebop.js:406:7)
    at Socket.emit (events.js:107:17)
    at readableAddChunk (_stream_readable.js:163:16)
    at Socket.Readable.push (_stream_readable.js:126:10)
    at TCP.onread (net.js:529:20)
```

zankich, 2015-06-23:
@timotius02 the `getMjpegStream` function only exists in the master branch and has not been released to npm yet. I'll push up a new version soon with the latest changes.

zankich, 2015-07-10:
@timotius02 version 0.3.0 has been released to npm and should resolve this issue.

closed 2015-07-10 by zankich