hybridgroup/node-bebop #6: bind EADDRINUSE

dk683, 2015-07-08:
Hello. I tried to run some sample code such as takeoff.js. 
However, I am consistently getting the following error:
```
dkkim930122@dkkim930122:~/node-bebop/examples$ node takeoff.js 
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: bind EADDRINUSE
    at exports._errnoException (util.js:746:11)
    at dgram.js:224:28
    at dns.js:85:18
    at process._tickCallback (node.js:355:11)
```
Could someone help me with this error?

For more information,
I installed node-bebop through using the following command:
npm install git:github.com/hybridgroup/node-bebop.git
My node version is v0.12.6 and npm is 2.11.2.

Thank you for your help!

deadprogram, 2015-07-08:
Hi, @dk683 are you sure you are connected from your computer to the drone's wifi? Remember you need to use the Bebop as an access point when connecting from your computer to the Bebop.

Or are you already connected from your mobile app to the drone? That error indicates some kind of IP address conflict, such as when you are already connected to the Bebop from something else.

Hope that helps!

dk683, 2015-07-08:
Hello @deadprogram! Thank you for your comment. 
Yes, I made sure that I connected my computer with Bebop. 
And the error was actually consistently happening after successfully flying (or getting mjpeg streaming) once. 

Previously, when I was working with Bebop's SDK code (https://github.com/Parrot-Developers/ARSDKBuildUtils), I noticed that they clean up Bebop(i.e. free reader thread array, stopping network) after commands were done.

However, if I noticed correctly, I could not find the clean up process in the nodejs code.
Could this be the reason why?

Thank you again for your help!

deadprogram, 2015-07-08:
That is not a problem we ourselves have run into ourselves. Node.js does not expose low-level threading routinues, and normally that is not needed. 

Can you provide the code sample, as well as letting us know what OS and version of Node you are using? Thanks.

dk683, 2015-07-08:
Thank you for your reply. 
The code sample of the SDK was: BebopDroneReceiveStream.c that can be found in the following link: https://github.com/Parrot-Developers/Samples/tree/master/Unix/BebopDroneReceiveStream.

My OS verison is Ubuntu 14.04 and my node version is v0.12.6 and npm is 2.11.2.

I will try to debug further and figure out what is exactly causing the connection problem! 

Thank you.

deadprogram, 2015-07-08:
I was referring to which node sample code you were running, sorry if that was unclear.

Do you have this same problem with other versions of node, such as 0.10.29?

dk683, 2015-07-08:
I am using the latest github node (https://github.com/hybridgroup/node-bebop), and the sample code was takeoff.js:

"use strict";

var bebop = require("../.");
```javascript
var drone = bebop.createClient();

drone.connect(function() {
  drone.takeOff();

  setTimeout(function() {
    drone.land();
  }, 5000);
});
```

I haven't tried the other versions of node yet, but I will try! 
Thank you.

dk683, 2015-07-10:
Dear @deadprogram,
Thank you! The problem was solved by downgrading node version from v0.12.6 to v0.10.29! :+1:

closed 2015-07-10 by dk683