hybridgroup/node-bebop #8: Takeoff Issue

dk683, 2015-07-17:
Hello. 
I tried the following code and noticed that takeoff was not working properly: it initially ascended for 0.5 meter and then descended near to a ground. And this behavior (up and down) was kept going on. I also tried the sample code of takeoff.js and changed setTiemout time to 20seconds, but the same problem happened. I would like to ask whether the same problem happened to other people.
Thank you for your time.

P.S. Just to make sure that the problem is on the hardware, I checked via taking off through app, but it worked well with the app.
 
```javascript
"use strict";

// Load modules
var bebop = require("../.")
var cv = require("opencv")

var drone = bebop.createClient()
var mjpg = drone.getMjpegStream()
var buf = null
var w = new cv.NamedWindow("Video", 0)

drone.connect(function() {

  // Drone is ready to operate
  drone.once("ready", function() {
    // Turn on mjepg streaming
    drone.takeoff() 

    drone.once("flying", function() {
      console.log("INFO: Drone take off success")

      loop = setInterval(function() {
        console.log("Do nothing")
      }, 100)
    })
  })
})
```

zankich, 2015-07-20:
@dk683 When you tested with the freeflight app, was the drone on the same surface as when you tried with your node program?

dk683, 2015-07-20:
@zankich Thank you for your reply. Yes, it was on the same surface (same room). Did you have the same problem what I had? Thank you!

zankich, 2015-07-20:
@dk683 which version of firmware does your drone have on it? I don't experience that issue with my drone. It takes off and lands just fine. Usually the drone will do that if there is an obstruction or if there is an uneven surface underneath the drone, but since the app works for you, it might be something in the API/firmware.

deadprogram, 2015-07-20:
Node version?

deadprogram, 2016-02-14:
Hi, @dk683 are you sill having troubles with the latest version of `node-bebop`? If so, please reopen this issue. Thanks!

closed 2016-02-14 by deadprogram