hybridgroup/node-bebop #9: Live stream to browser

akhilphilip, 2015-09-09:
Is there a way to live stream the h264 video received from getVideoStream() to an html <video> ?

johan-olsson, 2015-11-15:
I achieved this with [Broadway](https://github.com/mbebenita/Broadway)

deadprogram, 2015-11-16:
@johan-olsson how did you do it with Broadway? The README says `The video player first needs to download the entire video before it can start playing`. Can you please show a code sample? Thanks!

johan-olsson, 2015-11-16:
This is what I did. In this example I'm using Electron but you might as well send the base64 data over a socket or 206 response.

```javascript
var Drone = require('./Drone');
var Player = require('./Broadway/Player/Player');
var stream = Drone.getVideoStream();

var player = new Player({
  useWorker: true,
  workerFile: './lib/Broadway/Player/Decoder.js'
});

document.body.appendChild(player.canvas);

var toUint8Array = function (parStr) {
  var raw = atob(parStr);
  var array = new Uint8Array(new ArrayBuffer(raw.length));

  Array.prototype.forEach.call(raw, function (data, index) {
    array[index] = raw.charCodeAt(index);
  })

  return array;
};

stream.on('data', function (data) {
  player.decode(toUint8Array(data.toString('base64')));
});
```

johan-olsson, 2015-12-24:
@deadprogram did you get this working? In that case maybe we should close this issue and add a reference to the readme or something similar.

deadprogram, 2016-01-03:
I did not have a chance to test yet, but am charging some batteries now to check it out.

qwert789, 2016-03-03:
@johan-olsson I can confirm that the video not working with bebob 2 also other commands like land

Silvanosky, 2016-03-05:
http://developer.parrot.com/docs/bebop/#a-name-ardrone3-mediastreaming-videoenable-a
I think this is why people on bebop 2 can't get the stream

deadprogram, 2016-03-05:
@Silvanosky thanks for the link, sounds like we just need to call https://github.com/hybridgroup/node-bebop#mediastreamingvideoenableenable ala:

```
var bebop = require("node-bebop"),
      fs = require("fs");

var output = fs.createWriteStream("./video.h264"),
    drone = bebop.createClient(),
    video = drone.getVideoStream();

video.pipe(output);

drone.connect();
drone.MediaStreaming.videoEnable(1); // this line should start video
```

What do you think?

Silvanosky, 2016-03-05:
It would be better to call it in the connect callback to be sure the drone get the packet

deadprogram, 2016-03-06:
Good point. So that does seem to be the correct configuration call, we need to update the docs and examples.

qwert789, 2016-03-06:
@deadprogram, @Silvanosky MediaStreaming is null when doing drone.MediaStreaming.videoEnable(1);
so this is not working :(

Chetic, 2016-03-25:
I'm trying to see the video stream in some reasonable way where I don't need to see the drone to steer it. Is this even possible with node-bebop? The best solution I have so far is just doing "tail -f video.h264 | mplayer -" but we're still talking a second or two of lag and a LOT of decoding artifacts.

I've tried using your example @johan-olsson but I get an error saying "ReferenceError: print is not defined" in some minified code in Decoder.js:63. You can try it yourself at https://github.com/Chetic/beepbop

Anybody have a clue what I'm doing wrong? Is there a complete streaming example for node-bebop anywhere?

davibe, 2016-05-28:
i got the same error. Print os not defined (OSX)

fleish007, 2016-11-04:
not sure if this helps, but since the video is streaming on a different port you can include the stream port number (5504) as part of the discovery connection which will enable getting the video stream from vlc or other media players.

DA75, 2017-04-05:
@fleish007 What is the syntax of changing the port number ?