hybridgroup/node-bebop #10: Error loading file when running mjpeg-stream.js

ryanjbaxter, 2015-09-29:
I am trying to run the mjpeg-stream.js example but I am getting the error 

`[Error: Error loading file]`

Any idea why this is happening?  I am using node v0.12.4.

deadprogram, 2015-09-29:
Hi @ryanjbaxter pretty sure that version of node is not yet supported by this module? I cannot recall why at the moment. Can you please try an older version and see if you have same problem?

ryanjbaxter, 2015-09-29:
@deadprogram what version should i be using?

deadprogram, 2015-09-29:
I'd try 0.10.X

ryanjbaxter, 2015-09-29:
I switched to 0.10.38 and still see the same error.  I am OSX 10.10.5, not sure if that makes a difference.

ryanjbaxter, 2015-10-02:
Anyone have any other suggestions as to why this is not working?

deadprogram, 2015-10-02:
Do you have `ffmpeg` installed on your machine? I just realized the docs are missing the info that this is a needed prereq. The installation info for this on OSX  if you are using Homebrew is:

```
brew install ffmpeg
```

ryanjbaxter, 2015-10-02:
yes I have ffmpeg installed

nelson-liu, 2016-02-14:
ping @ryanjbaxter @deadprogram did you ever find a solution to this error?

ryanjbaxter, 2016-02-14:
@nelson-liu no I did not

deadprogram, 2016-04-15:
Hi, @ryanjbaxter can you please retry this code with the new 0.5.0 node-bebop release? I am not able to reproduce myself with this version. I will leave this issue open to await some additional info.

deadprogram, 2016-12-27:
Closing due to age of issue. Please reopen if needed.

closed 2016-12-27 by deadprogram