hybridgroup/node-bebop #11: Support For Getting GPS Coordinates

ryanjbaxter, 2015-10-06:
Is it possible to add support for getting the GPS coordinates of the drone?

deadprogram, 2015-10-07:
The Bebop API shows `PilotingStatePositionChanged` with a `PositionChanged` packet: https://github.com/Parrot-Developers/libARCommands/blob/master/Xml/ARDrone3_commands.xml#L539

So it *should* be possible.

ryanjbaxter, 2015-10-07:
Any pointers on how to implement that?  Maybe an example of an existing event that is already implemented in the node bebop library?

deadprogram, 2015-10-07:
This is the `FlyingStateChanged` packet code that is currently implemented:

https://github.com/hybridgroup/node-bebop/blob/master/lib/bebop.js#L241

deadprogram, 2016-02-14:
Hi, @ryanjbaxter you can now get the GPS info as part of the `PositionChanged` message. Check out https://github.com/hybridgroup/node-bebop/blob/master/examples/gps.js

ryanjbaxter, 2016-02-14:
Nice thanks I will take a look at it this week!  Has this been published to NPM?

deadprogram, 2016-02-14:
Yes, published yesterday.

deadprogram, 2016-04-15:
Hi @ryanjbaxter we've just released v0.5.0 which has many improvements including more support for the GPS events. Please give it a try, and if needed reopen this issue. Thanks!

closed 2016-04-15 by deadprogram