hybridgroup/node-bebop #13: Making it easier to send commands to drone.

johan-olsson, 2015-11-16:
I was thinking of implementing an easier way to communicate with the drone. My suggestion is that we use a similar method to getting the events from the xml files. 

For example if one would like to add a call to tilt/pan the camera:

First take a look in the xml to see what we need to do this.
```XML
    <class name="Camera" id="1">
        Ask the drone to move camera
        
        <cmd name="Orientation">
            Ask the drone to move camera.
            
            <arg name="tilt" type="i8">
                Tilt camera consign for the drone (in degree)
                The value is saturated by the drone.
                Saturation value is sent by thre drone through CameraSettingsChanged command.
            </arg>
            
            <arg name="pan" type="i8">
                Pan camera consign for the drone (in degree)
                The value is saturated by the drone.
                Saturation value is sent by thre drone through CameraSettingsChanged command.
            </arg>
        </cmd>
    </class>
```
Then call something like this:
```JavaScript
Drone.cmd(1, "Camera", "Orientation", 90, 0);
//Drone.cmd(<XMLFile>, <ClassName>, <CMDName>, ...<arguments>);
```

What do you think about this?

deadprogram, 2015-11-17:
That might work for an internal implementation. But the interface to the drone itself has 2 qualities I really want to preserve. 

One is as much compatibility with `node-ar-drone` as possible. This makes it possible to use other code that works with the current nodecopter ecosystem.

The other is having as appealing an API as possible. I'd much rather call something like `Drone.camera.orientation({tilt: 45, pan:45});`. What do you think?

johan-olsson, 2015-11-17:
Yeah, it was mostly meant for internal implementations. I just felt that it would be a lot easier then using the constants and writing the buffers by hand. I implemented some test functions the other day that look like this.
```javascript
Bebop.prototype.cameraOrientation = function(obj) {

  var buffer = commandToBuffer(1, "Camera", "Orientation", obj);

  this._writePacket(this._networkFrameGenerator(buffer));
  return this;
};

Bebop.prototype.commonAllStates = function() {

  var buffer = commandToBuffer(0, "Common", "AllStates");

  this._writePacket(this._networkFrameGenerator(buffer));
  return this;
};

Bebop.prototype.commonReboot = function() {

  var buffer = commandToBuffer(0, "Common", "Reboot");

  this._writePacket(this._networkFrameGenerator(buffer));
  return this;
};
```

deadprogram, 2015-11-17:
That looks very nice. Did you generate this code, or write by hand? Either way, I think you are on the right track.

deadprogram, 2015-11-17:
I'd love to see the commands namespaced, perhaps like `Bebop.camera.orientation()` or `Bebop.common.reboot()`.

johan-olsson, 2015-11-17:
So far I wrote these part by hand. But as you see it's very generateable so I was planning to do that just to get them all in there.

Sure I will just namespace them like that. I guess the only inconstancies then would be the once already there.
```javascript
Bebop.prototype.takeOff
```
would be the same as 
```javascript
Bebop.prototype.piloting.takeOff
```

deadprogram, 2015-11-17:
We should alias any commands to preserve the `node-ar-drone` commands, but if there are duplicates that would be ok, right? Programmers could just choose the newer API.

johan-olsson, 2015-11-18:
Agreed! Might even be a good idea to alias them eventually, just to save some processing.

deadprogram, 2016-02-16:
Now that this is in master branch, I am closing it :) Thanks again @johan-olsson for all your awesome work!

closed 2016-02-16 by deadprogram