hybridgroup/node-bebop #14: newbie question. how can i get videostream off the bebop

uaedronemaster, 2015-12-19:
Hello,

i wanted to know how can i get live video stream on an http local host?

and any references i can learn from? 

thanks

johan-olsson, 2015-12-24:
I think you will find this issue helpful #9

uaedronemaster, 2015-12-31:
Hello Johan,

I tried the code as you told me to but i am getting the following error.
may you please help me identify the source of the error.

[image: Inline image 1]

johan-olsson, 2015-12-31:
@uaedronemaster Think your image failed to upload or something.

uaedronemaster, 2015-12-31:
I havent installed emscripton thou, its not mandatory right! plus is there
an issue with the code ?

johan-olsson, 2015-12-31:
No it shouldn't be. But I meant that I can't see the image you posted of your error in your previous comment.

uaedronemaster, 2015-12-31:
> On Thu, Dec 31, 2015 at 8:47 PM, Adham Al-Khaja <admalkhaja@gmail.com>

well do u see it now?

uaedronemaster, 2015-12-31:
[image: Inline image 1]

uaedronemaster, 2015-12-31:
[image]

johan-olsson, 2015-12-31:
Does look like you haven't installed node-bebop. did you run `npm install node-bebop`?

uaedronemaster, 2015-12-31:
yes it is installed globally

uaedronemaster, 2015-12-31:
i installed it locally now it says 'Broadway-masterPlayerPlayer' module is
not found.... what is that

johan-olsson, 2015-12-31:
Install it locally and try again. You shouldn't install it globally

uaedronemaster, 2015-12-31:
Yes i did ... i got a new error now

uaedronemaster, 2015-12-31:
[image]

uaedronemaster, 2015-12-31:
I posted the new error picture

johan-olsson, 2016-01-01:
Maybe it's easier if you just post your code and I can have a look at it.

uaedronemaster, 2016-01-02:
this is the code:

var bebop = require('node-bebop');
var Player = require('./Broadway-master/Player/Player');
var Drone = bebop.createClient();
var stream = Drone.getVideoStream();

var player = new Player({
    useWorker: true,
    workerFile: './NodejsConsoleApp3/Broadway-master/Player/Decoder.js'
});

document.body.appendChild(player.canvas);

var toUint8Array = function (parStr) {
    var raw = atob(parStr);
    var array = new Uint8Array(new ArrayBuffer(raw.length));

    Array.prototype.forEach.call(raw, function (data, index) {
        array[index] = raw.charCodeAt(index);
    })

    return array;
};

stream.on('data', function (data) {
    player.decode(toUint8Array(data.toString('base64')));
});

johan-olsson, 2016-01-03:
Doesn't seem to be anything wrong with this assuming the paths are correct and you are creating an Electron app or similar. Is this the case?

uaedronemaster, 2016-01-07:
no i am not making and electron app.. what is that and am i suppose to make
an electron app to get the video live steam?

uaedronemaster, 2016-01-07:
I downloaded electron, but what should i do exactly? i am dragging and
dropping the visual studio .sln file in electron but it gives me an error?

deadprogram, 2016-01-07:
@uaedronemaster I would not worry about Electron at this point, that was I think just a general question.

uaedronemaster, 2016-01-07:
ah okay. but im not able to stream video yet ... help please

johan-olsson, 2016-01-07:
Alright, in that case you have to divide the code to backend nodejs and client js with Broadway. Something like this should work assuming you are using browserify for the client.

Node part:
```JavaScript
var io = require('socket.io')(80);
var bebop = require('node-bebop');
var Drone = bebop.createClient();
var stream = Drone.getVideoStream();

var toUint8Array = function (parStr) {
    var raw = atob(parStr);
    var array = new Uint8Array(new ArrayBuffer(raw.length));

    Array.prototype.forEach.call(raw, function (data, index) {
        array[index] = raw.charCodeAt(index);
    })

    return array;
};

io.on('connection', function (socket) {
  socket.emit('user connected');
stream.on('data', function (data) {
    player.decode(toUint8Array(data.toString('base64')));
});

});
```

Client part:
```JavaScript

var Player = require('./Broadway-master/Player/Player');

  var socket = io('http://localhost/');
  socket.on('connect', function () {
    socket.send('hi');

    socket.on('message', function (msg) {
      // my msg
    });
  });
```

johan-olsson, 2016-01-07:
@uaedronemaster I created a [project](https://github.com/johan-olsson/bebopBrowserVideo) for you to clone if you want an example on how to get it running.

uaedronemaster, 2016-01-08:
Thanks!! but just one last question... i ran the app.js using node and it
works and connects to the drone.... but where can i watch my live video
feed? i dont know where to type gulp-watch ..... i downloaded the chrome
extension which is the gulp-devtools but is that related ?

uaedronemaster, 2016-01-09:
Everything works well now including gulp watch... but the only problem is
that bufferutil module and utf8 cant be installed... it says error with
node-gyp rebuild ... cannot install optional dependencies. can you tell me
the version of nodejs and npm you are using ? thank you

johan-olsson, 2016-01-11:
What bufferutil and utf8 can't be installed? Could you show me your installation error?

uaedronemaster, 2016-01-12:
this is the error i get :

Microsoft Windows [Version 10.0.10586]
(c) 2015 Microsoft Corporation. All rights reserved.

[21:33:12] Using gulpfile ~\Desktop\Robotics\Parrot
Bebop\bebopBrowserVideo-master\gulpfile.js
[21:33:12] Starting 'html'...
[21:33:12] Starting 'broadway'...
[21:33:12] Starting 'fonts'...
[21:33:12] Finished 'fonts' after 77 μs
[21:33:12] Starting 'clean'...
[21:33:12] Finished 'clean' after 120 ms
[21:33:12] Starting 'moveCss'...
[21:33:12] Finished 'moveCss' after 1.84 ms
[21:33:12] Starting 'styles'...
[21:33:12] Finished 'styles' after 6.67 μs
[21:33:12] Starting 'scripts'...
[21:33:12] Starting 'bower'...
[21:33:12] Finished 'bower' after 4.62 μs
[21:33:12] all files 991 B
[21:33:12] Finished 'html' after 372 ms
[21:33:12] Finished 'broadway' after 189 ms
[BABEL] Note: The code generator has deoptimised the styling of
"C:/Users/Desktop/Robotics/Parrot
Bebop/bebopBrowserVideo-master/Broadway/Player/Decoder.js" as it exceeds
the max of "100KB".
[21:34:01] Browserify Error { [Error: Cannot find module 'utf-8-validate'
from 'C:\Users\Desktop\Robotics\Parrot
Bebop\bebopBrowserVideo-master\node_modules\ws\lib']
  stream:
   Labeled {
     _readableState:
      ReadableState {
        objectMode: true,
        highWaterMark: 16,
        buffer: [],
        length: 0,
        pipes: [Object],
        pipesCount: 1,
        flowing: true,
        ended: false,
        endEmitted: false,
        reading: true,
        sync: false,
        needReadable: true,
        emittedReadable: false,
        readableListening: false,
        defaultEncoding: 'utf8',
        ranOut: false,
        awaitDrain: 0,
        readingMore: false,
        decoder: null,
        encoding: null,
        resumeScheduled: false },
     readable: true,
     domain: null,
     _events:
      { end: [Object],
        error: [Object],
        data: [Function: ondata],
        _mutate: [Object] },
     _eventsCount: 4,
     _maxListeners: undefined,
     _writableState:
      WritableState {
        objectMode: true,
        highWaterMark: 16,
        needDrain: false,
        ending: true,
        ended: true,
        finished: true,
        decodeStrings: true,
        defaultEncoding: 'utf8',
        length: 0,
        writing: false,
        corked: 0,
        sync: false,
        bufferProcessing: false,
        onwrite: [Function],
        writecb: null,
        writelen: 0,
        bufferedRequest: null,
        lastBufferedRequest: null,
        pendingcb: 0,
        prefinished: true,
        errorEmitted: false },
     writable: true,
     allowHalfOpen: true,
     _options: { objectMode: true },
     _wrapOptions: { objectMode: true },
     _streams: [ [Object] ],
     length: 1,
     label: 'deps' } }
[21:34:01] Browserify Error { [Error: Cannot find module 'bufferutil' from
'C:\Users\Desktop\Robotics\Parrot
Bebop\bebopBrowserVideo-master\node_modules\ws\lib']
  stream:
   Labeled {
     _readableState:
      ReadableState {
        objectMode: true,
        highWaterMark: 16,
        buffer: [],
        length: 0,
        pipes: [Object],
        pipesCount: 1,
        flowing: true,
        ended: false,
        endEmitted: false,
        reading: true,
        sync: false,
        needReadable: true,
        emittedReadable: false,
        readableListening: false,
        defaultEncoding: 'utf8',
        ranOut: false,
        awaitDrain: 0,
        readingMore: false,
        decoder: null,
        encoding: null,
        resumeScheduled: false },
     readable: true,
     domain: null,
     _events:
      { end: [Object],
        error: [Object],
        data: [Function: ondata],
        _mutate: [Object] },
     _eventsCount: 4,
     _maxListeners: undefined,
     _writableState:
      WritableState {
        objectMode: true,
        highWaterMark: 16,
        needDrain: false,
        ending: true,
        ended: true,
        finished: true,
        decodeStrings: true,
        defaultEncoding: 'utf8',
        length: 0,
        writing: false,
        corked: 0,
        sync: false,
        bufferProcessing: false,
        onwrite: [Function],
        writecb: null,
        writelen: 0,
        bufferedRequest: null,
        lastBufferedRequest: null,
        pendingcb: 0,
        prefinished: true,
        errorEmitted: false },
     writable: true,
     allowHalfOpen: true,
     _options: { objectMode: true },
     _wrapOptions: { objectMode: true },
     _streams: [ [Object] ],
     length: 1,
     label: 'deps' } }

wrote:

johan-olsson, 2016-01-12:
I assume you have tried `npm install --save utf-8-validate bufferutil`? there might be some stuff that just don't come with node in windows. Never actually used it

uaedronemaster, 2016-01-12:
what do you use? mac? linux?

johan-olsson, 2016-01-12:
I use mac. But did you try to install the missing dependencies?

uaedronemaster, 2016-01-12:
Yes I did , but it fails at node-gyp rebuild

johan-olsson, 2016-01-17:
I really can't help you much with windows related issues. But the repository I linked you work so you should be able to google the more general errors you get

leerlemoniii, 2016-02-05:
little late to this, but the node-gyp issues are likely because you are using a newer version of node.... I find that using nvm-windows on a windows box allows me to install and change node version (and npm) which then fixes most these node-gyp problems.  Newer (v4 and 5) nodes have issues with nodegyp stuff that was compiled with the old 0. vaersions... When node and IO merged.

deadprogram, 2016-12-27:
Closing due to age of issue, and that the original point got a little lost. :) 

Please open new issues as needed, thanks everyone!

closed 2016-12-27 by deadprogram