hybridgroup/node-bebop #15: Camera movement?

strongautoconcepts, 2016-01-18:
Hi guys.

Haven't had a chance to dig through every line yet but so far I'm thoroughly impressed. 

One thing I didn't see was anything to do with moving the camera frame. I've noticed its mostly absent in all of the 3rd party bebop stuff. Is there a technical hurdle or is it just not implemented.

I appreciate all the hard work. Looking forward to participating where I can

deadprogram, 2016-01-18:
Hi, @strongautoconcepts thanks for the compliments. Actually, thanks to https://github.com/hybridgroup/node-bebop/pull/12 from @johan-olsson we are about to release that, along with *everything* else that the Bebop exposes in their public API.

strongautoconcepts, 2016-01-19:
fantastic!

thoughts on when #12 might be ready for public consumption?

closed 2016-01-19 by strongautoconcepts