hybridgroup/node-bebop #17: Not possible to get video stream

manuelh9r, 2016-02-12:
I want to fetch the video stream from the bebop 2. I tried the following code:

```
var drone = require('node-bebop');
var client = drone.createClient();
client.connect();
var video = client.getVideoStream();
video.on('data', console.log);
```

This throws the following error, and also the video example scripts are throwing the same error:

```
/home/mh9r/node_modules/node-bebop/lib/types.js:86
      return arg.enum[buffer.readInt32LE(index)]._name;
                                                ^

TypeError: Cannot read property '_name' of undefined
    at Object.module.exports.enum.read (/home/mh9r/node_modules/node-bebop/lib/types.js:86:49)
    at /home/mh9r/node_modules/node-bebop/lib/bebop.js:330:18
    at Array.forEach (native)
    at Bebop._packetReceiver (/home/mh9r/node_modules/node-bebop/lib/bebop.js:327:21)
    at emitTwo (events.js:100:13)
    at Socket.emit (events.js:185:7)
    at UDP.onMessage (dgram.js:481:8)
```

It seems that some parsing is not working correctly? The bebop 2 is on the latest firmware.

deadprogram, 2016-02-14:
Hi, @manuelh9r that fix was released as part of version 0.4.0 today. Please give it a try!

Silvanosky, 2016-02-22:
Hi, still having the problem : 

```
E:\Bebop\test\node_modules\node-bebop\examples>node video.js
E:\Bebop\test\node_modules\node-bebop\lib\types.js:86
      return arg.enum[buffer.readInt32LE(index)].name;
                                                ^
TypeError: Cannot read property 'name' of undefined
    at Object.module.exports.enum.read (E:\Bebop\test\node_modules\node-bebop\lib\types.js:86:49)
    at E:\Bebop\test\node_modules\node-bebop\lib\bebop.js:330:18
    at Array.forEach (native)
    at Bebop._packetReceiver (E:\Bebop\test\node_modules\node-bebop\lib\bebop.js:327:21)
    at Socket.emit (events.js:110:17)
    at UDP.onMessage (dgram.js:472:8)
```

johan-olsson, 2016-02-22:
What command are you using to get the stream?

Silvanosky, 2016-02-22:
I use the example video.js with the last version of node-bebop and a Bebop 2

```
var bebop = require("../."),
    fs = require("fs");

var output = fs.createWriteStream("./video.h264"),
    drone = bebop.createClient(),
    video = drone.getVideoStream();

video.pipe(output);

drone.connect();
```

johan-olsson, 2016-02-22:
@deadprogram do you think https://github.com/hybridgroup/node-bebop/blob/5d4c6dc4936fc8ae3bd4cb54c4f19a1e76412b19/lib/bebop.js#L298-L299 is to friendly? 
Video data should't even get there so I'll to see what kind of event it's trying to handle when I get home to the drone.

johan-olsson, 2016-02-23:
What version of node are you using? I have been trying quite a few versions now and I can't recreate your issue.

Silvanosky, 2016-02-23:
I have node v0.12.9

johan-olsson, 2016-02-23:
Hmm alright, just tested with that one too now. Still no issue. I don't have the bebop 2 but they should run the same software so it shouldn't be that. What OS are you running?

Silvanosky, 2016-02-23:
I'm on Windows

johan-olsson, 2016-02-23:
Maybe it's something with that then. I'm on OS X, let's hope @deadprogram can look a bit more in to this

Alain-Gateau, 2016-03-27:
I am having the same issue with the simple takeOff example script:

Users/alaingateau/node_modules/node-bebop/lib/types.js:86
      return arg.enum[buffer.readInt32LE(index)].name;
                                                ^
TypeError: Cannot read property 'name' of undefined
    at Object.module.exports.enum.read (/Users/alaingateau/node_modules/node-bebop/lib/types.js:86:49)
    at /Users/alaingateau/node_modules/node-bebop/lib/bebop.js:330:18
    at Array.forEach (native)
    at Bebop._packetReceiver (/Users/alaingateau/node_modules/node-bebop/lib/bebop.js:327:21)
    at emitTwo (events.js:87:13)
    at Socket.emit (events.js:172:7)
    at UDP.onMessage (dgram.js:481:8)

I am running node from a mac, with my bebop2 firmware updated to v 3.0.6

manuelh9r, 2016-04-04:
There seems to be an parsing/conversion error for the events sent from the drone (bebop.js:348). After a string value the offset is too large for a following enum value (offset must be 4 but is higher than 4).

With newer firmware the streaming disabled per default and needs to be enabled after connecting:
> Please note that on Bebop version less that 2.0.57 and Bebop 2 version less that 3.0.6 this command is not needed as the drone starts automatically the stream. To be fully compatible with future versions, you should in any cases send this command if you want to get the video stream.

deadprogram, 2016-04-04:
Hi @manuelh9r thanks for looking into this. There are several breaking changes with the 2.0.57 firmware that I am trying to track down. Now that Parrot removed the Github issues for all their repos, seems like one of us is going to have to dig in with Wireshark to figure out what the current protocol changes are, and what changes will be needed in node-bebop to accommodate them.

deadprogram, 2016-04-15:
Hi everyone we've just released v0.5.0 which has many improvements including fixes needed for the 3.x version of the Parrot API. Please give it a try, and if needed reopen this issue. Thanks!

closed 2016-04-15 by deadprogram