hybridgroup/node-bebop #19: unknown event output is undisplayable unicode (Mac OSX 10.11, english)

nelson-liu, 2016-02-13:
Hi,
I was testing out the GPS / networking functionalities, and noticed that the output when an unknown event is received is unable to be displayed on my computer. Here's how to reproduce:
````
$ node
> var bebop = require("path to bebop lib");
undefined
> var drone = bebop.createClient()
undefined
> drone.connect()
undefined
````
after this, it will immediately start logging:
````
recived unknown event �

recived unknown event �

recived unknown event �

recived unknown event �

recived unknown event �

recived unknown event �

recived unknown event �

recived unknown event �

recived unknown event �
````
is there an alternative to simply console.logging whenever an unknown network event is received? it quickly clogs up the log and makes interacting with the drone through the node interpreter impossible.

Thanks!

deadprogram, 2016-02-14:
Hi, @nelson-liu I just committed some code to emit an 'unknown' event, instead of just logging to the console. That should help address this issue.

closed 2016-02-14 by deadprogram

nelson-liu, 2016-02-14:
thanks, that indeed did!

commit 5d4c6dc4 (2016-02-14): Fix #19 by emitting 'unknown' event instead of just logging output