hybridgroup/node-bebop #20: No Events fired

TheOssi, 2016-03-01:
I registered to several events (ready,battery,landed,takingOff, Video, etc) but when running my application the only event that is fired is the ready event. All other events seem not be emitted or something. Even the example events.js is not working for me.
I installed the newest firmware on the drone and also got the latest version of node-bebop. Taking off and piloting works fine.

Any suggestions?

deadprogram, 2016-03-01:
Hi, @TheOssi thanks for reporting. What is your exact firmware version?

TheOssi, 2016-03-01:
2.0.57 is the firmware version of the bepop

johan-olsson, 2016-03-02:
Are the "Raw Events" firing? `BatteryStateChanged`, `FlyingStateChanged` etc?

TheOssi, 2016-03-02:
@johan-olsson unfortunately they are not firing either.

deadprogram, 2016-03-03:
@TheOssi can you please post your code? Thanks.

TheOssi, 2016-03-03:
```
var bebop = require('node-bebop');

var drone = bebop.createClient();

drone.connect(function () {
    drone.takeOff();

    setTimeout(function () {
        drone.land();
    }, 5000);
    drone.on("navData", function (data) {
        console.log("Alt changed: " + data);
    });
    drone.on("ready", function () {
        console.log("ready");
        console.log(drone.navData);
    });

    drone.on("battery", function (data) {
        console.log(data);
    });

    drone.on("landed", function () {
        console.log("landed");
    });

    drone.on("takingOff", function () {
        console.log("takingOff");
    });

    drone.on("hovering", function () {
        console.log("hovering");
    });

    drone.on("FlyingStateChanged", function () {
        console.log("FlyingStateChanged");
    });

    drone.on("BatteryStateChanged", function () {
        console.log("BatteryStateChanged");
    });

    drone.on("takingOff", function() {
        console.log("takingOff");
    });

    drone.on("flying", function() {
        console.log("flying");
    });

    drone.on("landing", function() {
        console.log("landing");
    });

    drone.on("unknown", function(data) {
        console.log("unknown", data);
    });
});
```

deadprogram, 2016-03-04:
Hi @TheOssi I made sure my Bebop 1 had firmware 2.0.57. I connected to the Wifi, and ran your provided code using Node 4.2.2 under Ubuntu 14.04 and it ran fine: https://gist.github.com/deadprogram/d54df9e08508a6b8099e

What Node.js version and OS are you using?

TheOssi, 2016-03-04:
Hi @deadprogram i am usually running node 5.7.1, but now I downgraded it to 4.2.2 for testing but it did not work either. My OS is windows 7.

TheOssi, 2016-03-04:
Update: Apparently my McAfee Firewall was somehow blocking the port for the incoming data from the drone. I turned it off and it works fine now.

closed 2016-03-04 by TheOssi