hybridgroup/node-bebop #21: Moving Speed is inverted

TheOssi, 2016-03-04:
I was testing out a bit of the data and functions and recognized that the moving speed is actually negative when moving forward, even tho it is documented as postive(http://developer.parrot.com/docs/bebop/#a-name-ardrone3-pilotingstate-speedchanged-a).
This caused me some problems until I figured it out.
So is this intended or should it not be inverted?

deadprogram, 2016-03-05:
That sounds like an error to me, we should follow the API as written.

deadprogram, 2016-04-28:
Hi @TheOssi are you still having this problem with the latest release?

TheOssi, 2016-04-29:
Hi @deadprogram im currently not working anymore with the bepop, I only had it for a limited time. So i can't check it at the moment. Sorry :)

CE93Geek, 2017-03-14:
@TheOssi Hi, do you remember what is the unit of the speed parameter?