hybridgroup/node-bebop #22: Drone starts drifting

vsafin-copart, 2016-03-05:
It seems like once I make drone take off using drone.take Off() and then start doing any other movements clockwise or right and left it start s drifting a lot all over the place. I am wondering if I am missing something

Program is really straight forward with takeoff and clockwise wrapped into setTimeout just like in the dance.js example I would expect drone to stay in place but it drifts away for at few meters in the matter of 10-15 sec

deadprogram, 2016-03-06:
Hi @vsafin-copart you should check the firmware version on the drone. Also, you might want to use https://github.com/hybridgroup/node-bebop#speedsettingshullprotectionpresent or https://github.com/hybridgroup/node-bebop#speedsettingsoutdooroutdoor depending on your environment.

xliefde, 2016-04-02:
im having the same problem, the firmware version on the drone is up to date (2.0.57)

deadprogram, 2016-04-02:
Quite a few people seem to have had problems with drift on the Bebop, their forums have a lot of threads like https://community.parrot.com/t5/Bebop-Drone/Bebop-started-drifting-again/m-p/131673/highlight/true#M13193

I suggest running the calibration (via the iOS or Android app), and see if that helps out. I do not personally recommend doing firmware downgrades, as a couple people recommended on that thread, simply because it introduced different problems without really solving any for me (at least at that point).

xliefde, 2016-04-02:
The strange thing is that when i use the free flight app on my phone ,it's not drifting ,it only starts drifting when i'm using the API

deadprogram, 2016-04-02:
I had a similar problem a couple times, and I cannot recall offhand what I did about it. Thinking...

deadprogram, 2016-04-09:
Hi, everyone

I pushed a new branch https://github.com/hybridgroup/node-bebop/tree/bugfix/control-pcmd that handles the PCMD flag differently. Still testing it today, but please feel free to give it a try and let how it works for you.

deadprogram, 2016-04-15:
Hi everyone we've just released v0.5.0 which has many improvements including greatly improved stability when hovering. Please give it a try, and if needed reopen this issue. Thanks!

closed 2016-04-15 by deadprogram