hybridgroup/node-bebop #23: Video stream is 'jumpy'

Sporks, 2016-03-11:
I noticed that when recording or streaming video from my Parrot drone, the video stream does not seem to be consistent.  It's almost as if it stops randomly and then starts again.  Any ideas why this might be?

deadprogram, 2016-03-12:
Hello, @Sporks are you referring to the recording on the drone itself, or recording on your computer from the stream coming from the drone?

Sporks, 2016-03-21:
I am referring to the video stream to the computer, when using .getVideoStream()

deadprogram, 2016-03-22:
The quality of that stream is highly variable, by design I think. Parrot is trying to reduce latency at the cost of quality, and hence they recommend that recording is best done on the drone itself. 

That said, perhaps there may be some things that could improve on the client side. I'll take another look.

gcwelborn, 2016-03-25:
GIF image of video: 
http://gcwelborn.github.io/img/drone-dashboard-example.gif

@deadprogram Maybe you can confirm that this is as good as it gets. 
The frames themselves are very high quality, but the framerate constantly fluctuates.

deadprogram, 2016-03-25:
That GIF is very cool looking... except for the latency, of course. :)

My own machine's performance looks a lot better, however, there are several factors that might be impacting. I am often using OpenCV, rather then streaming to a web page with the additional weight of the mjpeg encoding/decoding.

Any way you can share the code with me, so I can try to reproduce your same setup?

Also, what OS/Node version/etc are you using?

gcwelborn, 2016-03-25:
Thanks @deadprogram. I first attempted OpenCV with no such luck, then moved on to Broadway - which @Sporks was able to get working. Important to note it's an electron application - and the Broadway folder has been gitignored, let me if you encounter any issues.

The code: https://github.com/gcwelborn/drone-dashboard
OS: OSX 10.11.3 (Macbook Pro mid-2010, 4gb RAM)
Node:  v5.5.0

deadprogram, 2016-04-15:
Hi, @gcwelborn while reading thru the Parrot dev forums, I saw a couple interesting tidbits. Seems that when the PCMD pings sent to the Parrot are not being received as often as it expects, the Parrot responds by throttling the video. Several devs mentioned to make sure to use the 5.8ghz wifi as well.

ChrisWilliamson123, 2016-08-12:
Hi, apologies for this being such a delayed post, but I only started working with this repo about 2 weeks ago.

I was having a similar problem to this so I commented out lines 696-707 here and set emit to true on line 694.
https://github.com/hybridgroup/node-bebop/blob/master/lib/bebop.js#L696

That stopped the issue of the video being 'jumpy' but adds some quality issues (I feel that this way round is easier to manage). I'm assuming this change means that the code doesn't wait for a full frame or something before sending the next one.

Hope this helps.