hybridgroup/node-bebop #27: I'm trying to set the drone to indoor flight mode but it's throwing value out of bounds error

xliefde, 2016-04-02:
var bebop = require('node-bebop');

var drone = bebop.createClient();

drone.connect(function() {
 drone.SpeedSettings.outdoor(0);
});

> 
> TypeError: value is out of bounds
>     at TypeError (<anonymous>)
>     at checkInt (buffer.js:784:11)
>     at Buffer.writeUInt8 (buffer.js:794:5)
>     at Object.module.exports.u8.write (/home/esra/node_modules/node-bebop/lib/types.js:9:14)
>     at /home/esra/node_modules/node-bebop/lib/commandToBuffer.js:59:22
>     at Array.forEach (native)
>     at writeBuffer (/home/esra/node_modules/node-bebop/lib/commandToBuffer.js:57:11)
>     at module.exports (/home/esra/node_modules/node-bebop/lib/commandToBuffer.js:68:10)
>     at SpeedSettings.outdoor (/home/esra/node_modules/node-bebop/lib/SpeedSettings.js:37:16)
>     at /home/esra/Projects/AVA/test.js:6:22
>

deadprogram, 2016-04-02:
I suspect that this line of code here https://github.com/hybridgroup/node-bebop/blob/master/lib/SpeedSettings.js#L37 should probably be:

```
var buffer = commandToBuffer(0, "SpeedSettings", "Outdoor", outdoor);
```

I'm charging up my batteries to try this out.

deadprogram, 2016-04-02:
I just tried the unmodified code, and did not have that problem. What version of Node are you running?

xliefde, 2016-04-02:
i'm running v0.10.25

deadprogram, 2016-04-02:
Hi, @xliefde I tested using both Node 5.8.0 and 4.2.2 and had no problems. What OS are you running? I tested with Ubuntu 14.04.

xliefde, 2016-04-02:
That's weird i'm running Ubuntu 14.04 too 
The firmware on the drone is up to date ( 2.0.57)

deadprogram, 2016-04-02:
OK, was able to repro you problem in Node v0.10.40

I'll take a look to see if there is a simple solution for backward compat with old versions of Node.

deadprogram, 2016-04-02:
I think this is similar to a problem solved by @johan-olsson previously, perhaps he can be persuaded to look into it.

deadprogram, 2016-04-28:
Hi @xliefde are you still having this problem with the latest release?

xliefde, 2016-04-29:
Unfortunately still it's same and my motherboard is broken now. even if i use the ios app for drone. It's still drifting.. After i fix it i'll work on sdk which is python.

deadprogram, 2016-04-30:
Hi @xliefde I've just confirmed that the new updates are working in Node 0.10.40, 4.2.2, and 5.8.0

I am going to close this issue, please reopen if you need further assistance when you get your drone fixed. Thanks again for your help!

closed 2016-04-30 by deadprogram