hybridgroup/node-bebop #28: I need to send my bebop 2 to a GPS point

brunoua, 2016-04-06:
HI,

I nedd to send my bebop 2 to a GPS coordinate. Can i use this command "
GPSSettings.sendControllerGPS(obj)"? I try this but nothing happens.

Regards,
Bruno Barbosa

deadprogram, 2016-04-06:
Hi, @brunoua the `sendControllerGPS` command is to tell the drone the location of the user for returning home during autonomous flight, not to tell the drone to navigate to that GPS location. According to Parrot docs:

```
Send controller GPS location

    Send controller GPS location:

deviceController->aRDrone3->sendGPSSettingsSendControllerGPS(deviceController->aRDrone3, (double)latitude, (double)longitude, (double)altitude, (double)horizontalAccuracy, (double)verticalAccuracy);

Send controller GPS location

    latitude (double): GPS latitude in decimal degrees
    longitude (double): GPS longitude in decimal degrees
    altitude (double): GPS altitude in meters
    horizontalAccuracy (double): Horizontal Accuracy in meter ; equal -1 if no horizontal Accuracy
    verticalAccuracy (double): Vertical Accuracy in meter ; equal -1 if no vertical Accuracy

The user location might be used in case of return home, according to the home type and the accuracy of the given position.
```

brunoua, 2016-04-06:
Hi, @deadprogram  the only way I have is with mavlink?

Thanks for available time

deadprogram, 2016-04-06:
That is my understanding, yes.

deadprogram, 2016-04-15:
Hi @brunoua we've just released v0.5.0 which has many improvements including the Mavlink stuff you were such a great help with. Please give it a try, and if needed reopen this issue. Thanks!

closed 2016-04-15 by deadprogram