hybridgroup/node-bebop #29: Compatibility  with bebop 1 ?

py4object, 2016-04-09:
Hey i'm wondering what should i do to make this api compatible with be
currently it's working but the drone is hella drifting and hard to control

deadprogram, 2016-04-09:
Hi, @py4object I pushed a branch named https://github.com/hybridgroup/node-bebop/tree/bugfix/control-pcmd that seems to fly with greater stability for me with the 2.0.57+ firmware.

Please give it a try and see how it works for you.

I also added a new alias command `drone.level();` for the poorly named `drone.stop()` to reset the PCMD flag to 0, which is what I think the root cause for the stability problem with the latest firmware.

deadprogram, 2016-04-15:
Hi @py4object we've just released v0.5.0 which has many improvements including the stability one I previously mentioned. Please give it a try, and if needed reopen this issue. Thanks!

closed 2016-04-15 by deadprogram

py4object, 2016-04-17:
Hi sorry ,I didn't have the time to try it out 
Now that i just did ,it's flying really stable ! cheers

deadprogram, 2016-04-18:
Glad to hear it! :helicopter:

CE93Geek, 2017-03-28:
@py4object hi, what firmware did you have on bebop1 when it was stable?