hybridgroup/node-bebop #30: getVideoStream() nor getMjpegStream() Is not working after upgrade to 3.9.21

xliefde, 2016-04-20:
Hi after i've upgraded the drone to 3.9.21 getVideoStream or getMjpegStream dont seem to be working 

Here is a sample of my code 

```
var bebop = require('node-bebop');

var drone = bebop.createClient();
drone.connect();
 stream = drone.getVideoStream()

 drone.MediaStreaming.videoEnable(1);

 stream.on('data',function (data) {
	 console.log(data)
	
})

```

xliefde, 2016-04-20:
Just FYI if someone is having same problem now the stream is by default disabled (which wasn't the case ) you have to enable it on connection

deadprogram, 2016-04-20:
Hi @xliefde thanks for the feedback. You are absolutely correct, according to the Parrot docs, all versions of 3.x+ firmware have is disabled by default. 

We had already previously modified all the examples to now have the needed `drone.MediaStreaming.videoEnable(1);` call. Do you think there should also be some mention in the README? If so, would you be willing to submit a PR with this info?

Thanks again.

oscar619, 2017-07-20:
Hi @xliefde Sorry to bother you, you have some updated information to visualize the video of the drone parrot bebop 2?