hybridgroup/node-bebop #32: Still error with arg.enum[value].name node_modules/node-bebop/lib/types.js:102 name = arg.enum[value].name;

kuche1991, 2016-04-29:
I used the standard example and still got

> node_modules/node-bebop/lib/types.js:102
>         name = arg.enum[value].name;
>                         
>
>TypeError: Cannot read property 'name' of undefined
>    at Object.module.exports.enum.read (/home/kuche/git/node_modules/node-bebop/lib/types.js:102:33)
>    at Bebop._packetReceiver (/home/kuche/git/node_modules/node-bebop/lib/bebop.js:343:16)
>    at emitTwo (events.js:87:13)
>    at Socket.emit (events.js:172:7)
>    at UDP.onMessage (dgram.js:480:8)

 with Version 0.5.0.

Fixed it in types.js line 99-108 temporary with:

```
...
if (!arg.enum || value > arg.enum.length) {
          name = "unknown enum";
        } else {
          try {
              name = arg.enum[value].name;
          }catch(e){
              console.log("Exception in fix @types.js/105");
              name = "unknown enum";
          }
        }
...
```

Maybe there is a better Solution...

deadprogram, 2016-04-29:
Sorry, which one is the "standard example"? :)

Also, what OS/version of Node/Bebop firmware are you running?

kuche1991, 2016-04-29:
Sorry, i meant the one in the "How to use"-Section:

```
var bebop = require('node-bebop');

var drone = bebop.createClient();

drone.connect(function() {
  drone.takeOff();

  setTimeout(function() {
    drone.land();
  }, 5000);
});
```

We are using Node 4.4.3 on Ubuntu.
Our drone is the Bebop 2 with Firmware-Version 3.2.0 (the currently latest i think).

deadprogram, 2016-04-30:
Hi @kuche1991 

I have Ubuntu 14.04. I installed Node 4.4.3 using nvm, and deleted/reinstalled my node modules.

I have a Bebop 1,  also running the 3.2.0 firmware. Ran that code without any problem.

That particular code is called when some event occurs that is supposed to be described by the enum. In this case it seems to be an unknown value. I cannot match your exact conditions, however I can code a bit more defensively to handle it.

I've just made a commit to the master branch that should address this situation, and hopefully your case.

Can you give it a try with your setup, please?

kuche1991, 2016-04-30:
Hi @deadprogram,

thanks for you quick response!
I won't have access to our drone over the weekend, but as soon as I have, I will tryout your solution and report the outcome.

kuche1991, 2016-05-02:
Hi @deadprogram,

today i was able to test this with the drone again and all my tests worked out well now :)
Thanks again for your quick response!

closed 2016-05-02 by kuche1991