hybridgroup/node-bebop #34: Upgraded Firmware to 3.2.0 on BB Now can't connect

bcrosby333, 2016-05-14:
I get a EHOSTUNREACH 192.168.1.102:44444 when i attempt to create the client now.

Are there any other steps needed to enable node-bebop with a v3.2.0 firmware on the BB?

deadprogram, 2016-05-14:
Hi, @bcrosby333 I am running same firmware no such problem, and so are many others, so it not some inherent problem. 

I would suggest your verify that your computer is connected to the same WiFi as the drone provides, before running your node-bebop code.

Hope that helps!

bcrosby333, 2016-05-14:
@deadprogram I am connecting to a AP via script after it starts up. See https://github.com/Parrot-Developers/Samples/wiki/Convert-Bebop-to-client-instead-of-access-point 

There is where it fails. So i am not sure if it has anything to do with the node-bebop code. B/c you are right that if i just connect to the Bebop directly it works fine. 

Any thoughts?

deadprogram, 2016-05-14:
If you know the IP address of the Bebop, you can override in your code by using the `ip` option

```
var drone = bebop.createClient({ip: "YOUR REAL IPADDRESS GOES HERE"});
```

bcrosby333, 2016-05-15:
Yep, that is how i connect. However it doesn't work when you modify the settings on the Bebop to connect to Wifi Access Point via it being the Access Point and you connecting to it.

deadprogram, 2016-05-15:
It might be obvious, but you need to enter your bebop's IP address, not your computer's IP address.

How are you determining your bebop's IP address, by looking at your router?

One other thing to check, make sure your router is not setup to disallow connects between machines on the internal subnet, the way many hotel/conference wifi is setup..

Can I please see a sample of your code?

deadprogram, 2016-12-26:
Since this appears to be something with the user's WiFi config and not node-bebop itself, I am closing. Thanks!

closed 2016-12-26 by deadprogram