hybridgroup/node-bebop #40: Example code 'mjpeg-stream.js' isn't working...

tzist, 2016-07-28:
Hello,

I'm trying to run an example code 'mjpeg-stream.js', but I can't get images.

I checked this code, and I found variant buf's value is still 'null'.

I guess a Bebop Drone's image data was not transferred to variant 'data'.

mjpg.on("data", function(data) {
  **buf = data;                                     ->  The data was not transferred. I guess...**
});

What should I do????

Somebody help me

SairyLyms, 2016-09-22:
The problem is same as on my Bebop2(Firmware ver. 3.3.0)
I think the problem is caused by this issue:
http://forum.developer.parrot.com/t/cannot-decode-video-after-upgrade-sdk-from-3-6-to-3-8-3/

deadprogram, 2016-12-26:
Hi, everyone. For sure that is not the problem, as I am running firmware 3.3.0 as well. I just tested again to be certain, and it worked.

I suggest you look into your opencv installation, as that is required to use this example.

I am going to close this issue, please reopen if needed. Thanks!

closed 2016-12-26 by deadprogram