hybridgroup/node-bebop #41: Take Picture and Send to server?

clokam, 2016-08-26:
Is there anyway that we can use takePicture and send it back to the server. We are controlling the drone, but want to be able to take a picture from the drone and send it to a visual recognition API so if we can get it back to the server where we are controlling the drone we can do the rest.

deadprogram, 2016-08-26:
Hi @clokam you could certainly use that API call, and then subsequently use the FTP protocol to download the image file to the computer from the drone. Of course, you would need to either use a bridge or change the drone to connect in infrastructure mode to be able to access both the drone and the Internet connection from the computer.

Hope that helps!

clokam, 2016-08-26:
I am kind of new to this library and hence was slightly confused. Once we take the picture, how do we identify from internal memory that it was the picture taken just now. What sort of data structure is used to store the image?

Secondly, is there any documentation around the FTP protocol for the drone and how we can enable infrastructure mode so that we are connected to the drone and the Internet?

Your help is much appreciated!

deadprogram, 2016-12-26:
The images are just normal jpg files. There is no special info about the FTP server on the drone, it is also just normal FTP.

Hope that helps!

closed 2016-12-26 by deadprogram