hybridgroup/node-bebop #44: video

Sramir27, 2016-11-11:
Having trouble grabbing the live video stream on my mac. Very new to all of this. was wondering if someone could explain a stepwise process on grabbing the video.

deadprogram, 2016-12-26:
Hi, @Sramir27 there is not a step by step guide anywhere, sorry. The assumption is you will know how to work with the h.264 encoded video stream. I would suggest you investigate tools like ffmpeg.

You can run the `video.js` example like this:

```
node examples/video.js
```

which will create a file named `video.h264` with the streamed data from the drone. Then in a separate console, you can open a playback window using ffplay (part of ffmpeg):

```
ffplay ./video.h264
```

Hope that helps!

closed 2016-12-26 by deadprogram

Skorkmaz88, 2016-12-26:
I am having issue with h264 file it is size zero and ffplay did not work. System is bebop 2 with firmware 4.0

deadprogram, 2016-12-26:
@Skorkmaz88 are you connected to the drone'e wifi network from your computer? Can you run other samples such as https://github.com/hybridgroup/node-bebop/blob/master/examples/takeoff.js and https://github.com/hybridgroup/node-bebop/blob/master/examples/events.js ?