hybridgroup/node-bebop #45: CurrentDate and CurrentTime commands not working.

ChrisWills123, 2016-11-13:
Hi, the code I am executing to try and change the date are as follows:

```
var today = new Date().toISOString().split('T');
drone.Common.currentDate(today[0]);
drone.Common.currentTime('T' + today[1]);
```

I've inspected the packet that is sent to the drone and compared it with what is sent from the phone app (FreeFlight 3).

```
Date, phone -> drone
04 0b 01 16 00 00 00 00 04 01 00 32 30 31 36 2d 31 31 2d 31 33 00

Date, computer -> drone
02 0a 04 15 00 00 00 00 04 01 00 32 30 31 36 2d 31 31 2d 31 33
```

From bebop.py I can see that the first 2 bytes correspond to the data type and id. How come these are different? Shouldn't they be the same as FreeFlight uses the official SDK?

deadprogram, 2016-11-16:
Hi, @ChrisWills123 there appear to have been a number of breaking changes to the Parrot API in that release. The XML definition file has even moved to a different repo https://github.com/Parrot-Developers/arsdk-xml

I will try to look into this when I get a chance to get my Bebop out and flash it to the latest firmware. In the meantime, any help to resolve would be greatly appreciated.

ChrisWills123, 2016-11-17:
Hi @deadprogram

Yeah I've even tried downgrading the drone but not really had any luck there.

I'll take a deeper look into it tomorrow and see if I can resolve any of the issues (this one and the video stuff).

Cheers for the response.
Chris

deadprogram, 2016-12-26:
Hi, @ChrisWills123 

There are some changes here https://github.com/hybridgroup/node-bebop/tree/feature/firmware-4.0 that should correct your issues. Also check out the sample https://github.com/hybridgroup/node-bebop/blob/feature/firmware-4.0/examples/setdatetime.js strangely you must set both date and time for the drone to fire off the `CurrentDateChanged` and `CurrentTimeChanged` events.