hybridgroup/node-bebop #46: getVideoStream not working with Bebop 4.0 firmware

ChrisWills123, 2016-11-13:
Hi, for some reason, updating to 4.0 has stopped getVideoStream from working.

I've checked for the video packets in WireShark too and they are just not coming through from the drone.

deadprogram, 2016-11-16:
Hi @ChrisWills123 please see my comment on your other issue, basically there appear to have been a number of breaking changes to the Parrot API in that release.

arthursp, 2016-12-22:
Any news about this ? Will you implement it on a new version ?

deadprogram, 2016-12-26:
Hi, everyone. Sorry about delayed response.

The latest Parrot firmware that I tested (3.3.0 on my Bebop 1) the default is now to not stream video.

The solution is to call the `MediaStreaming` interface's `videoEnable()` function with a value of `1`:

`drone.MediaStreaming.videoEnable(1);`

Several of the examples already are doing this, because just calling `getVideoStream()` just obtains the stream object, but does not tell the drone to start streaming.

I do not have a Bebop 2 to test the 4.0 firmware, however I would expect the same behavior based on my reading of the Parrot SDK files. The last version released for Bebop 1 is currently 3.3.0.

In any case, I am working on updating the API signatures to include some of the newer API calls to add support for the newer Parrot drones such as the Disco.

I am closing this issue as solved, please reopen if needed.

closed 2016-12-26 by deadprogram

Skorkmaz88, 2016-12-26:
This means video example should work out of the box right?

deadprogram, 2016-12-26:
Yes, but If for some reason not working for you, you might want to check out the https://github.com/hybridgroup/node-bebop/tree/feature/firmware-4.0 branch which has the latest commands.

ChrisWilliamson123, 2017-02-22:
Hi @deadprogram 

The video.js example doesn't seem to work with the Bebop 2 even with this new firmware branch. The drone records video to storage but the stream is non-existent.

I ran the video.js example but nothing was output to the video.h264 file. If you get change please could you look into this. If there's anything I can do to help just let me know.

Cheers,
Chris