hybridgroup/node-bebop #47: Unexpected NULs in flightpath filename

Phrohdoh, 2016-12-06:
Following https://github.com/hybridgroup/node-bebop/blob/88512f1bfa0e8321c269961442a373fa4c9e1df5/examples/flightplan.js I am running into some issues.

I have tried both filepaths with the same results:

```javascript
  drone.on("AvailabilityStateChanged", function(data) {
    console.log("AvailabilityStateChanged", data);
    console.log(data.AvailabilityState);
    if (data.AvailabilityState === 1 && !alreadyFlying) {
      alreadyFlying = true;
      //drone.Mavlink.start("/data/ftp/internal_000/flightplans/much-smaller-flightplan.mavlink", 0);
      drone.Mavlink.start("much-smaller-flightplan.mavlink", 0);
    }
  });
```

```
ready
BatteryStateChanged
76
unknown <Buffer 00 05 0b 00 00 0e 00 0e 00 00 00 00 00>
FlyingStateChanged
landed
ComponentStateListChanged { component: 'Calibration', State: 1 }
ComponentStateListChanged { component: 'GPS', State: 1 }
ComponentStateListChanged { component: 'Mavlink_File', State: 1 }
ComponentStateListChanged { component: 'TakeOff', State: 1 }
AvailabilityStateChanged { AvailabilityState: 1 }
1
MavlinkFilePlayingStateChanged { state: 'stopped',
  filepath: '/data/ftp/internal_000/flightplans/much-smaller-flightplan.mavlink\u0000\u0000\u0000\u0000\u0000',
  type: 'unknown enum' }
GPSFixStateChanged { fixed: 1 }
unknown <Buffer 01 14 07 00 01 00 00 00>
unknown <Buffer 01 14 08 00 01 00 00 00>
unknown <Buffer 01 14 09 00 00 00 00 00>
unknown <Buffer 86 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00>
unknown <Buffer 86 00 04 00 01 00 00 00 3f 00 7f 00>
unknown <Buffer 86 00 04 00 02 00 00 00 00 00 7f 00>
unknown <Buffer 86 00 04 00 03 00 00 00 00 00 7f 00>
MavlinkFilePlayingStateChanged { state: 'playing',
  filepath: '\u0000\u0000\u0000\u0000\u0000',
  type: 'invalid enum' }
ComponentStateListChanged { component: 'Mavlink_File', State: 1 }
ComponentStateListChanged { component: 'TakeOff', State: 1 }
ComponentStateListChanged { component: 'Mavlink_File', State: 0 }
AvailabilityStateChanged { AvailabilityState: 0 }
0
MavlinkFilePlayingStateChanged { state: 'stopped',
  filepath: '/data/ftp/internal_000/flightplans/much-smaller-flightplan.mavlink\u0000\u0000\u0000\u0000\u0000',
  type: 'unknown enum' }
unknown <Buffer 01 04 09 00 26 a3 7c 2e 40 6e 40 40 4e 16 0c ae 39 3c 58 c0 00 00 00 00 00 b0 60 40 00 00 00>
unknown <Buffer 01 04 09 00 26 8d e2 2f 40 6e 40 40 13 0c bf ae 39 3c 58 c0 00 00 00 00 00 b0 60 40 00 00 00>
unknown <Buffer 01 04 09 00 26 61 ae 32 40 6e 40 40 70 00 72 af 39 3c 58 c0 00 00 00 00 00 b0 60 40 00 00 00>
unknown <Buffer 01 04 09 00 66 77 48 31 40 6e 40 40 70 00 72 af 39 3c 58 c0 00 00 00 c0 cc ac 60 40 00 00 00>
```

I don't know why there are NULs in my filepath (this is probably resulting in my `invalid enum` type):
```
MavlinkFilePlayingStateChanged { state: 'stopped',
  filepath: '/data/ftp/internal_000/flightplans/much-smaller-flightplan.mavlink\u0000\u0000\u0000\u0000\u0000',
  type: 'unknown enum' }

MavlinkFilePlayingStateChanged { state: 'playing',
  filepath: '\u0000\u0000\u0000\u0000\u0000',
  type: 'invalid enum' }
```

Could I get some help regarding exactly which filepath I should give it and where there are null bytes being appended to the paths?

Phrohdoh, 2016-12-06:
I even tried 'fixing' `Mavlink.js` with no change.

```javascript
Mavlink.prototype.start = function(filepath, type) {
  var filename = filepath;
  if (!filename.endsWith("\u0000"))
    filename += "\u0000";

  console.log("!! new filename !!: " + filename);
  var buffer = commandToBuffer(0, "Mavlink", "Start", filename, type);

  this.self._writePacket(this.self._networkFrameGenerator(buffer));
  return this.self;
};
```

deadprogram, 2016-12-26:
Hi @Phrohdoh sorry about delayed response.

Not sure about your exact problem, but the Parrot API returns some extra null characters, but this was not impacting my use of autopilot. As of this commit https://github.com/hybridgroup/node-bebop/commit/1bb86af1f4b53e0eeb44ed46abe653fc979dff71 my flightplan was able to run, once my Bebop obtained a GPS signal.

I would suggest you look very closely at your mavlink flight plan file. I created mine using the Parrot Flightplan software, and moved it around using FTP. Also note that the path in question is a path on the drone itself, not on your local machine.

Hope that helps!

deadprogram, 2016-12-26:
Oops, did not mean to close.

deadprogram, 2016-12-26:
@Phrohdoh this is the long saga of how we got mavlink working originally: https://github.com/hybridgroup/node-bebop/issues/25

Phrohdoh, 2016-12-26:
Thank you! I was hoping to not have to purchase Parrot's software (F/OSS preference) but will do so if necessary.

I won't have any time to work on this project today but should hopefully sometime this week.
I'll bug you again if something goes strange or I'll close this if I am able to resolve my issue.

Thanks again! :-)