hybridgroup/node-bebop #49: keyboard control

Skorkmaz88, 2016-12-27:
Sorry it is not really a problem, I just wonder what would be the best approach to have control with the keyboard keys using this module. Currently, RTP example for firmware 4 seems to be working fine for video access, ideally it should be combined with video stream

deadprogram, 2016-12-27:
I'd personally suggest using https://github.com/hybridgroup/cylon-bebop and https://github.com/hybridgroup/cylon-keyboard

Kind of like https://github.com/hybridgroup/cylon-bebop/blob/master/examples/bebop_dualshock3.js but replacing the joystick with the keyboard.

Skorkmaz88, 2016-12-27:
yet, do they support, firmware 4 ?

deadprogram, 2016-12-27:
cylon-bebop is a wrapper around node-bebop.

Skorkmaz88, 2016-12-27:
cylon-bebop's new version causing a jump  before landing and acting weirdly, not able to land. It used work better. (Bebop 2, firmware 2), maybe I should add this issue to there

deadprogram, 2016-12-27:
Good idea, please do so. But I think it is using older node-bebop, likely, so probably that is the problem, it needs an update to the next node-bebop.

Skorkmaz88, 2016-12-27:
Done so, I think this issue can be closed

closed 2016-12-27 by deadprogram