hybridgroup/node-bebop #53: moving commands speed parameter

CE93Geek, 2017-03-14:
Hi,
I would like to know what is the unit used in implementing the speed parameter in the moving functions like "forward", "backward", "left" and "right". It says in the documentation that the speed are from 0 to 100 but what it is in term of what? like meter/sec? I need this information urgently as I am trying to make bebop moves in certain distance.

regards,

johan-olsson, 2017-03-15:
I think it's the amount of forward tilt the drone has. Pretty sure you change it in the settings too but I guess that doesn't help you much. Can't you use the gps?

CE93Geek, 2017-03-15:
@johan-olsson Hi, I didn't get what do mean by "forward tilt" could you please clarify it more? I really appreciate it. And for the GPS no I can't use it because in our project we are restricted to indoor experiments and the characteristic of our work is to plan autonomous navigation based on image processing (track target).

johan-olsson, 2017-03-15:
I actually meant pitch I guess :) [Aircraft_principal_axes](https://en.wikipedia.org/wiki/Aircraft_principal_axes). If you can't use GPS maybe this event is helpful [SpeedChanged](https://github.com/hybridgroup/node-bebop/blob/master/docs/events.md#speedchanged)

CE93Geek, 2017-03-15:
@johan-olsson I think you are right, I was reading /lib/bebop.js file and the forward function documented as the change in the pitch value somehow.
Thank you very much for your response.