hybridgroup/node-bebop #59: mjpeg_stream.js not emitting data buffers

erangiar, 2017-06-08:
hello I'm totally new to drone programming and i'm trying to run the examples provided with my bebop 2. so far I've been able to send basic commands(eg : takeoff.js, dance.js) but I'm having a problem in receiving data from the drone. for example when I try to run mjpeg_stream.js it only prompts a empty window but no data. I've successfully installed opencv and ffmpeg packages. when i tested the code i found that following function doesn't trigger.

mjpg.on("data", function(data) {
  buf = data;  ------->>> (here the data is still null)
});  

Is there any pre-configuration to be done in order to receive the stream from the drone? Please someone help me to get this issue solved. thanks in advance.

the versions I'm using
node v6.10.0
npm  v3.10.10
firmware v.4.0.5 
os - win 10

pedroma2s, 2017-06-13:
Hi, I have same exact problem and so far I'm not been able to solve it nor determine its cause. If I may ask, did you upgraded the drones firmware recently?

erangiar, 2017-06-13:
@pedroma2s yes I upgraded to firmware v.4.0.5 
I've even tried the examples in firmware 4.0 folder as well. but didn't work yet

pedroma2s, 2017-06-13:
@erangiar I had v.4.0.5 and it worked fine until I upgraded to the v.4.0.6. I rolled back the upgrades to the v.3.3.0 and then v.4.0.5, the version that I had before, in neither of those versions worked again. So I don't suggest wasting your time in trying that.
Seems that the problem has something to do with the upgrade process and not with the firmware version. I also have an open issue regarding this problem but my prayers have not been answered, and at this point I have not much hope left.

erangiar, 2017-06-18:
@pedroma2s worked fine means did it give the drone camera feed via the opencv window? Did you do any kind of pre configurations like changing port numbers or something. I'm asking just because I'm very new to this node js environment :) And have you tried opencv face tracking example?

pedroma2s, 2017-06-20:
@erangiar actually a was a streaming the images to a python module and then displaying them using opencv. I didn't do any pre configurations. And no, I have not tried the opencv face tracking example, just a toturial to detect objects of certain colors.
[image]
Here is the code I was using to send the images.