hybridgroup/node-bebop #60: Which Bebop Firmware version compatible is this code, Video Streaming?

ahnHeejune, 2017-08-31:
Which version of bebop firmware should I use to use video protocol? 

In fact, I used ok with this program last year, but after I upgraded to 4.0 it does not work.  I do not remember which version I used last year on my Bebop. So I tried with 3.3 version but it doe snot work with your code.

Can you tell me which one should I use?  Please tel me

Skorkmaz88, 2017-08-31:
I was able to retrieve video from bebop 2 with framework version 4 around the beginning of this year. Actually, it didn't work for me before upgrading to 4.0

ahnHeejune, 2017-08-31:
Thanks for your reply. 
1) what you mean by "framework" ? you mean firmware?  
2) In fact, i am using Bebop-1 
3) As far as i know, the video frame format that you use is as follows:

type (1B) :  Id (1B): seq (1B): length (4B): fn (2B): 
flag (1B): fragnum (1b) : num_fragments_per_frame (1B)
video data (variable size)
for example)
03:		7d:		4d:	f4:03:00:00:	02:00:	00:	0a:		0e:  

But with 4.0, Drone doesnot send the this format any more. ^^;;