hybridgroup/node-bebop #65: Error bebop.js :  Cannot read property 'readUInt8' of undefined

kolki, 2018-01-13:
i have encountered this error

TypeError: Cannot read property 'readUInt8' of undefined
    at Bebop._packetReceiver (/home/kit/node_modules/node-bebop/lib/bebop.js:301:43)
    at emitTwo (events.js:87:13)
    at Socket.emit (events.js:172:7)
    at UDP.onMessage (dgram.js:480:8)

as i test any command.

in fact i modify bebop.js like this 

line 307 FROM

```
 if ( networkFrame.id === constants.BD_NET_DC_EVENT_ID || networkFrame.id === constants.BD_NET_DC_NAVDATA_ID){

```
TO

```
 if ( (networkFrame.id === constants.BD_NET_DC_EVENT_ID ||
  networkFrame.id === constants.BD_NET_DC_NAVDATA_ID) && networkFrame.data !==  undefined) 
```

it seems sometime data field have lengfh but is empty.

```
{ type: 4,
  id: 126,
  seq: 53,
  size: 15,
  data: <Buffer 01 04 0f 00 03 00 00 00> }
{ type: 4, id: 126, seq: 54, size: 7 }
/home/kit/node_modules/node-bebop/lib/bebop.js:307
    var commandProject = networkFrame.data.readUInt8(0),
                                          ^
```

arapocket, 2018-04-30:
Keep getting this as well

Starystars67, 2018-05-09:
I too am getting this error at the moment

Starystars67, 2018-05-09:
Making this edit to /lib/bebop.js allows things to work without this readUInt8 but i dont know what issues it might cause down the line:

`if (networkFrame.id === constants.BD_NET_DC_EVENT_ID ||
      networkFrame.id === constants.BD_NET_DC_NAVDATA_ID) {

    //var //commandProject = networkFrame.data.readUInt8(0),
        //commandClass = networkFrame.data.readUInt8(1),
        //commandId = networkFrame.data.readUInt16LE(2);

    var offset = 4;
    var args = {};
    var event = null;

    try {
      event = commands
        .first({id: commandProject}).class
        .first({id: commandClass}).cmd;
    }
    catch (err) {
      this.emit("unknown", networkFrame.data);
    }

    if (event) {

      if (event instanceof Array) {
        event = event[commandId];
        if (!event) {
          this.emit("unknown", networkFrame.data);
          return;
        }
      }

      if (typeof event.arg !== "undefined") {
        if (event.arg instanceof Array) {
          event.arg.forEach(function(arg) {
            if (types.hasOwnProperty(arg.type)) {
              args[arg.name] = types[arg.type]
                .read(networkFrame.data, offset, arg);

              offset += types[arg.type].length;
            }
          });
        }
        else if (event.arg instanceof Object) {
          if (types.hasOwnProperty(event.arg.type)) {
            args[event.arg.name] = types[event.arg.type]
              .read(networkFrame.data, offset, event.arg);
          }
        }
      }

      this.emit(event.name, args);
    }

    /*switch (commandProject) {
      case constants.ARCOMMANDS_ID_PROJECT_COMMON:
        switch (commandClass) {
          case constants.ARCOMMANDS_ID_COMMON_CLASS_COMMONSTATE:
            switch (commandId) {
              case constants.ARCOMMANDS_ID_COMMON_COMMONSTATE_CMD_BATTERYSTATECHANGED:
                this.navData.battery = networkFrame.data.readUInt8(4);
                this.emit("battery", this.navData.battery);
                break;
            }
            break;
        }
        break;
      case constants.ARCOMMANDS_ID_PROJECT_ARDRONE3:
        switch (commandClass) {
          case constants.ARCOMMANDS_ID_ARDRONE3_CLASS_PILOTINGSTATE:
            switch (commandId) {
              case constants.ARCOMMANDS_ID_ARDRONE3_PILOTINGSTATE_CMD_FLATTRIMCHANGED:
                break;
              case constants.ARCOMMANDS_ID_ARDRONE3_PILOTINGSTATE_CMD_FLYINGSTATECHANGED:
                switch (networkFrame.data.readInt32LE(4)) {
                  case constants.ARCOMMANDS_ARDRONE3_PILOTINGSTATE_FLYINGSTATECHANGED_STATE_LANDED:
                    this.navData.flyingState = { landed: true };
                    this.emit("landed");
                    break;
                  case constants.ARCOMMANDS_ARDRONE3_PILOTINGSTATE_FLYINGSTATECHANGED_STATE_TAKINGOFF:
                    this.navData.flyingState = { takingOff: true };
                    this.emit("takingOff");
                    break;
                  case constants.ARCOMMANDS_ARDRONE3_PILOTINGSTATE_FLYINGSTATECHANGED_STATE_HOVERING:
                    this.navData.flyingState = { hovering: true };
                    this.emit("hovering");
                    break;
                  case constants.ARCOMMANDS_ARDRONE3_PILOTINGSTATE_FLYINGSTATECHANGED_STATE_FLYING:
                    this.navData.flyingState = { flying: true };
                    this.emit("flying");
                    break;
                  case constants.ARCOMMANDS_ARDRONE3_PILOTINGSTATE_FLYINGSTATECHANGED_STATE_LANDING:
                    this.navData.flyingState = { landing: true };
                    this.emit("landing");
                    break;
                  case constants.ARCOMMANDS_ARDRONE3_PILOTINGSTATE_FLYINGSTATECHANGED_STATE_EMERGENCY:
                    this.navData.flyingState = { emergency: true };
                    this.emit("emergency");
                    break;
                }
                break;
            }
            break;
        }
        break;
    }*/
  }`

evilqubit, 2018-08-11:
so what is a permanent fix ?

alishibli97, 2019-01-08:
Did you find any solution?? please