hybridgroup/node-bebop #69: VideoEnabledChanged returns error at Bebop drone

Shtivi, 2018-05-12:
Hi,
i've been trying to get the video stream from my bebop (1) drone, according to the guide:
https://github.com/hybridgroup/node-bebop/blob/master/examples/video.js.
But the stream is not starting and I don't get the 'video', 'data' or 'error' events emitted.
I tried to debug it with listening to the VideoEnabledChanged event and when I sends the enableVideo command it sends error.
Do you have any idea?

arthursp, 2018-05-31:
Yes : the bepop sdk has changed and this repo didn't follow up

Shtivi, 2018-06-03:
Humm.. ok, so what do I have to do to get it work?

arthursp, 2018-06-04:
Two options : 

- Take the C SDK from parrot, and reverse engineering the process of coding + decoding packets
- Use Wireshark to see which command is send when streaming video from official app